AndreiBarsan/DynSLAM #33: compile issue

rnunziata, 2017-09-22:
I was getting the following bug

Track.cpp:73:30: error: ‘setw’ was not declared in this scope

but fixed it by adding the line at the top of the file.

#include <iomanip>

Now it fails with following errors....not sure what to do?

```
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/VisoSparseSFProvider.cpp: In function ‘Eigen::Matrix4d instreclib::VisoToEigen(const Matrix&)’:
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/VisoSparseSFProvider.cpp:9:28: error: passing ‘const Matrix’ as ‘this’ argument discards qualifiers [-fpermissive]
   return Eigen::Matrix4d((~viso_matrix).val[0]);
                            ^
In file included from /home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/../../libviso2/src/matcher.h:34:0,
                 from /home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/SparseSFProvider.h:7,
                 from /home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/VisoSparseSFProvider.h:7,
                 from /home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/VisoSparseSFProvider.cpp:3:
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/../../libviso2/src/matrix.h:106:11: note:   in call to ‘Matrix Matrix::operator~()’
   Matrix  operator~ ();                // transpose
           ^
[ 30%] Building CXX object src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/PrecomputedSegmentationProvider.cpp.o
src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/build.make:158: recipe for target 'src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/VisoSparseSFProvider.cpp.o' failed
make[2]: *** [src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/VisoSparseSFProvider.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/PrecomputedSegmentationProvider.cpp: In member function ‘virtual std::shared_ptr<instreclib::segmentation::InstanceSegmentationResult> instreclib::segmentation::PrecomputedSegmentationProvider::SegmentFrame(const Mat3b&)’:
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/PrecomputedSegmentationProvider.cpp:173:25: error: ‘imread’ is not a member of ‘cv’
   cv::Mat seg_preview = cv::imread(img_fpath);
                         ^
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/PrecomputedSegmentationProvider.cpp:174:3: error: ‘resize’ is not a member of ‘cv’
   cv::resize(seg_preview, *last_seg_preview_, cv::Size(), 1.0 / input_scale_, 1.0 / input_scale_, cv::INTER_LINEAR);
   ^
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/DynSLAM/InstRecLib/PrecomputedSegmentationProvider.cpp:174:99: error: ‘INTER_LINEAR’ is not a member of ‘cv’

```

AndreiBarsan, 2017-09-22:
Do you have OpenCV installed on your system? (If so, perhaps there's some sort of version clash. I will have to double check which exact version is needed.)

Otherwise, maybe try adding `#include <opencv/cv.h>` to the problematic file(s).

Apologies for the vague responses, it's just that I currently do not have access to a system with an nVIDIA GPU and CUDA so that I can test these things.

rnunziata, 2017-09-23:
I have installed opencv 3.3, however I think I can get around this...its the Eigen error that I am not sure what to do about.

AndreiBarsan, 2017-09-26:
Hmm, so the `return Eigen::Matrix4d((~viso_matrix).val[0]);` is causing an error? Again, I can't try out any changes right now (really sorry!), but can you try making a (non-const) copy of `viso_matrix`, transposing it (using the `~` operator), and using `val[0]` to construct an Eigen matrix?

rnunziata, 2017-09-26:
I would ...however I am not a c++ programmer and not sure how formulate
this statement into code. Can you supply the replacement line: both before
and after  lines. I would be glad to test it.

rnunziata, 2017-09-27:
closed see above fix

closed 2017-09-27 by rnunziata