AndreiBarsan/DynSLAM #34: compile errors.

rnunziata, 2017-09-27:
compile errors....have way though compile....should I download new version of InfinitTAM

```
[ 44%] Building CXX object src/InfiniTAM/InfiniTAM/Engine/CMakeFiles/Engine.dir/LibUVCEngine.cpp.o
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/InfiniTAM/InfiniTAM/Engine/LibUVCEngine.cpp: In function ‘void callback_depth(uvc_frame_t*, void*)’:
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/InfiniTAM/InfiniTAM/Engine/LibUVCEngine.cpp:57:42: error: ‘UVC_FRAME_FORMAT_GRAY16’ was not declared in this scope
  data->framebuffer_depth->frame_format = UVC_FRAME_FORMAT_GRAY16;
                                          ^
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/InfiniTAM/InfiniTAM/Engine/LibUVCEngine.cpp: At global scope:
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/InfiniTAM/InfiniTAM/Engine/LibUVCEngine.cpp:62:8: error: ‘uvc_frame_desc’ does not name a type
  const uvc_frame_desc *frame_desc;
        ^
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/InfiniTAM/InfiniTAM/Engine/LibUVCEngine.cpp: In function ‘FormatSpec findSuitableFormat(uvc_device_handle_t*, int, int, const char*)’:
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/InfiniTAM/InfiniTAM/Engine/LibUVCEngine.cpp:79:6: error: ‘struct FormatSpec’ has no member named ‘frame_desc’
  ret.frame_desc = NULL; ret.interval = 0;
      ^
/home/rjn/opencv3-p3-code/classification_and_boxing/newstuff/DynSLAM/src/InfiniTAM/InfiniTAM/Engine/LibUVCEngine.cpp:81:8: error: ‘uvc_format_desc_t’ does not name a type
  const uvc_format_desc_t *formats = uvc_get_format_descs(dev);

```

rnunziata, 2017-10-07:
fixed by adding 
#include <iomanip>

closed 2017-10-07 by rnunziata

AndreiBarsan, 2017-10-13:
Very sorry for replying so slowly; I still haven't had the chance to make an install guide or to access a machine with CUDA. :(