AndreiBarsan/DynSLAM #37: Link error....how to include pthread

rnunziata, 2017-10-08:
Link error....how to include pthread . Can someone help

```
[ 84%] Linking CXX executable InfiniTAM
/usr/bin/ld: ITMLib/libITMLib.a(ITMMainEngine.cpp.o): undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/InfiniTAM/InfiniTAM/CMakeFiles/InfiniTAM_cli.dir/build.make:110: recipe for target 'src/InfiniTAM/InfiniTAM/InfiniTAM_cli' failed
make[2]: *** [src/InfiniTAM/InfiniTAM/InfiniTAM_cli] Error 1
CMakeFiles/Makefile2:873: recipe for target 'src/InfiniTAM/InfiniTAM/CMakeFiles/InfiniTAM_cli.dir/all' failed
make[1]: *** [src/InfiniTAM/InfiniTAM/CMakeFiles/InfiniTAM_cli.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: ITMLib/libITMLib.a(ITMMainEngine.cpp.o): undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/InfiniTAM/InfiniTAM/CMakeFiles/InfiniTAM.dir/build.make:110: recipe for target 'src/InfiniTAM/InfiniTAM/InfiniTAM' failed
make[2]: *** [src/InfiniTAM/InfiniTAM/InfiniTAM] Error 1
CMakeFiles/Makefile2:914: recipe for target 'src/InfiniTAM/InfiniTAM/CMakeFiles/InfiniTAM.dir/all' failed
make[1]: *** [src/InfiniTAM/InfiniTAM/CMakeFiles/InfiniTAM.dir/all] Error 2
[100%] Built target DynSLAMGUI
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2
```

rnunziata, 2017-10-08:
added -lpthread to InfinitTam CMakeLists ..

closed 2017-10-08 by rnunziata

amiltonwong, 2017-11-20:
Hi, @rnunziata ,

Could you elaborate how to modify the file `DynSLAM/src/InfiniTAM/InfiniTAM/CMakeLists.txt` to have the function (added -lpthread)?

THX!

rnunziata, 2017-11-20:
around line 75 in CMakeList.txt

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}  -pthread")

amiltonwong, 2017-11-20:
Thanks @rnunziata

AndreiBarsan, 2017-11-24:
Thanks for chiming in @rnunziata !!