AndreiBarsan/DynSLAM #38: Have you considered make this a ROS node

rnunziata, 2017-10-12:
Have you considered make this a ROS node so that depth, and segment classification can be integrated via ROS messages. This would allow switching between different implementation of these components, and would be language agnostic.

AndreiBarsan, 2017-10-13:
Yes, I have. You are correct; those would be very neat advantages, but it would take a bit of time to refactor everything. It's on my todo list, though. And once again, very sorry about the slow replies, it's a busy period here and I'm also CUDA-less. :(

AndreiBarsan, 2017-10-13:
Note to self: Peidong suggested ROS2 should be used (http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29).

0dmf0, 2020-04-27:
> Yes, I have. You are correct; those would be very neat advantages, but it would take a bit of time to refactor everything. It's on my todo list, though. And once again, very sorry about the slow replies, it's a busy period here and I'm also CUDA-less. :(

Have you already make this a ROS node?

AndreiBarsan, 2020-04-30:
Sorry, I haven't added ROS support and I don't think I'll have time in the foreseeable future. I'd be more than happy to review PRs adding that if someone does it!

0dmf0, 2020-05-01:
Do you know how to make a KITTI data with a rosbag which has rgb and depth message?

发自我的iPhone

AndreiBarsan, 2020-05-01:
ETH Zurich seem to have a utility for building rosbags from KITTI sequences here: https://github.com/ethz-asl/kitti_to_rosbag

However, since KITTI doesn't come with depth (apart from the LiDAR), you'd need to compute that separately, and modify the code above to include one additional depth or disparity image in each frame.

If you want to, you can use the DispNet also used by this project, or if you want to upgrade to newer tech (DispNet is almost 5 years old), I'd recommend one of my colleague's awesome DeepPruner: https://github.com/uber-research/DeepPruner (MIT licensed)

0dmf0, 2020-05-02:
> ETH Zurich seem to have a utility for building rosbags from KITTI sequences here: https://github.com/ethz-asl/kitti_to_rosbag
>
> However, since KITTI doesn't come with depth (apart from the LiDAR), [...]

Well, so can you explain clearly how to use another KITTI dataset? I don't know how to process another dataset.

0dmf0, 2020-05-02:
Or can you provide another KITTI dataset which has already been processed?
Thanks

AndreiBarsan, 2020-05-02:
Do you mean pre-processed for DynSLAM or pre-processed for ROS? 
For DynSLAM there are tools in this repo (MNC and dispnet-flownet-docker in `preprocessing/`) do do it for KITTI data. For ROS I am not sure; I am not familiar with ROSbags.

0dmf0, 2020-05-02:
> Do you mean pre-processed for DynSLAM or pre-processed for ROS?
> For DynSLAM there are tools in this repo (MNC and dispnet-flownet-docker in `preprocessing/`) do do it for KITTI data. For ROS I am not sure; I am not familiar with ROSbags.

Everytime I pre-process KITTI data, it shows:
dmf@dmf-GS63-7RE:/media/dmf/丁/DynSLAM$ scripts/preprocess-sequence.sh kitti-tracking data/kitti/tracking training 6Failure: Invalid directory structure for the tracking dataset. 
Expected directory structure:
		├── training
		│   ├── calib
		│   ├── image_02
		│   │   ├── 0000
		│   │   └── etc.
		│   ├── image_03
		│   │   ├── 0000
		│   │   └── etc.
		│   └── velodyne
		│       ├── 0000
		│       └── etc.
		└── testing
			└── <same as training>.
Do you know how to solve this?

AndreiBarsan, 2020-05-02:
What does your dir structure look like?

0dmf0, 2020-05-02:
[image]
is it wrong?

0dmf0, 2020-05-02:
/media/dmf/丁/2020-05-02 23-05-35屏幕截图.png

AndreiBarsan, 2020-05-02:
Seems like the folder is called `training 1` instead of `training`. Maybe
that's the issue?

0dmf0, 2020-05-02:
bash:./preprocess-sequence.sh: no such file
Do you know why?