AndreiBarsan/DynSLAM #40: build error related to eigen3

amiltonwong, 2017-11-18:
Hi, all,

I encounter the following build error 

```
[ 55%] Building CXX object CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/Evaluation/EvaluationCallback.cpp.o
[ 56%] Building CXX object CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DSHandler3D.cpp.o
In file included from /data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp:2:0:
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:15:10: error: ‘Eigen’ does not name a type
   static Eigen::Vector3d GetTranslation(const pangolin::OpenGlRenderState &stat
          ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:23:10: error: ‘Eigen’ does not name a type
   static Eigen::Vector3d GetDirection(const pangolin::OpenGlRenderState &state)
          ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:28:10: error: ‘Eigen’ does not name a type
   static Eigen::Vector3d GetEuler(const Eigen::Matrix3d &rot) {
          ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:77:3: error: ‘Eigen’ does not name a type
   Eigen::Vector3d eye;
   ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:78:3: error: ‘Eigen’ does not name a type
   Eigen::Vector3d direction;
   ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h: In constructor ‘dynslam::gui::DSHandler3D::DSHandler3D(pangolin::OpenGlRenderState*, pangolin::AxisDirection, float, float)’:
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:37:7: error: class ‘dynslam::gui::DSHandler3D’ does not have any field named ‘eye’
     : eye(GetTranslation(*cam_state)),
       ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:37:36: error: ‘GetTranslation’ was not declared in this scope
     : eye(GetTranslation(*cam_state)),
                                    ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:38:7: error: class ‘dynslam::gui::DSHandler3D’ does not have any field named ‘direction’
       direction(GetDirection(*cam_state)),
       ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:38:40: error: ‘GetDirection’ was not declared in this scope
       direction(GetDirection(*cam_state)),
                                        ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:45:5: error: ‘Eigen’ has not been declared
     Eigen::Matrix4d mv = cam_state->GetModelViewMatrix();
     ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:46:5: error: ‘Eigen’ has not been declared
     Eigen::Vector3d euler = GetEuler(mv.block(0, 0, 3, 3));
     ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.h:48:25: error: ‘euler’ was not declared in this scope
     yaw_accum_ = euler(1) + M_PI_2;
                         ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp: In member function ‘void dynslam::gui::DSHandler3D::UpdateModelViewMatrix()’:
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp:23:3: error: ‘direction’ was not declared in this scope
   direction = rot_quat * Eigen::Vector3d(1.0f, 0.0f, 0.0f);
   ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp:25:28: error: ‘eye’ was not declared in this scope
   Eigen::Vector3d target = eye + direction;
                            ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp: In member function ‘virtual void dynslam::gui::DSHandler3D::MouseMotion(pangolin::View&, int, int, int)’:
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp:45:36: error: ‘direction’ was not declared in this scope
       Eigen::Vector3d newMotionX = direction.cross(up_v).normalized() * dx * 0.
                                    ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp:47:7: error: ‘eye’ was not declared in this scope
       eye += newMotionX;
       ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp: In member function ‘virtual void dynslam::gui::DSHandler3D::Mouse(pangolin::View&, pangolin::MouseButton, int, int, bool, int)’:
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp:94:7: error: ‘eye’ was not declared in this scope
       eye += direction.normalized() * zoom_scale_;
       ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp:94:14: error: ‘direction’ was not declared in this scope
       eye += direction.normalized() * zoom_scale_;
              ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp:97:7: error: ‘eye’ was not declared in this scope
       eye -= direction.normalized() * zoom_scale_;
       ^
/data/code/DynSLAM/src/DynSLAM/DSHandler3D.cpp:97:14: error: ‘direction’ was not declared in this scope
       eye -= direction.normalized() * zoom_scale_;
              ^
CMakeFiles/DynSLAMGUI.dir/build.make:350: recipe for target 'CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DSHandler3D.cpp.o' failed
make[2]: *** [CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DSHandler3D.cpp.o] Error 1
CMakeFiles/Makefile2:74: recipe for target 'CMakeFiles/DynSLAMGUI.dir/all' failed
make[1]: *** [CMakeFiles/DynSLAMGUI.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

```

In my system, I already install eigen3 by:
`sudo apt-get install libeigen3-dev`

Any suggestion to fix this problem?

THX!

AndreiBarsan, 2017-11-20:
I can't check this approach right now, but have you tried fiddling with `/cmake/FindEigen3.cmake`. Perhaps it's not finding Eigen correctly in the location the package manager sets it up. Personally, I've never used the apt version of Eigen, and instead build the project using the version of Eigen included as a git submodule. Have you tried CMaking the version of Eigen inside the submodule?

amiltonwong, 2017-11-20:
Thanks @AndreiBarsan , it works when Eigen it built internally.
THX!

closed 2017-11-20 by amiltonwong

AndreiBarsan, 2017-11-24:
Yay! Please let me know if you have other questions (apart from the separate issue you opened, to which I just replied) and I'll do my best to answer. Actually, at the moment I don't have access to a machine with an nVidia GPU on which to test the system properly (and make it easier to build, etc.), but I will do my best to answer your questions to the best of my abilities.

amiltonwong, 2017-11-24:
Thanks a lot :)  @AndreiBarsan