AndreiBarsan/DynSLAM #41: about usage

amiltonwong, 2017-11-20:
Hi, @AndreiBarsan ,

Thanks for sharing such a good work. And now I'm practicing it.
After I had built it, I follow steps mentioned in readme:
Download the kitti seq (2011_09_26_drive_0001 synced+rectified) from here
>> cd DynSLAM/build
>> ./DynSLAMGUI  --dataset_root=/data/kitti_raw_data_seq/2011_09_26/2011_09_26_drive_0001_sync

```
root@milton-OptiPlex-9010:/data/code/DynSLAM/build# ./DynSLAMGUI  --dataset_root=/data/kitti_raw_data_seq/2011_09_26/2011_09_26_drive_0001_sync
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not open calibration file: [/data/kitti_raw_data_seq/2011_09_26/2011_09_26_drive_0001_sync/calib.txt]
Stack trace (most recent call last):
#11   Object ", at 0xffffffffffffffff, in 
#10   Object "./DynSLAMGUI, at 0x452e56, in _start
#9    Source "/build/eglibc-oGUzwX/eglibc-2.19/csu/libc-start.c", line 287, in __libc_start_main [0x7f9bb7c9af44]
#8    Source "/data/code/DynSLAM/src/DynSLAM/DynSLAMGUI.cpp", line 1307, in main [0x44d71d]
       1305:   dynslam::DynSlam *dyn_slam;
       1306:   dynslam::Input *input;
      >1307:   BuildDynSlamKittiOdometry(dataset_root, &dyn_slam, &input);
       1308: 
       1309:   dynslam::gui::PangolinGui pango_gui(dyn_slam, input);
       1310:   pango_gui.Run();
#7    Source "/data/code/DynSLAM/src/DynSLAM/DynSLAMGUI.cpp", line 1163, in dynslam::BuildDynSlamKittiOdometry(std::string const&, dynslam::DynSlam**, dynslam::Input**) [0x48bdcd]
       1160:   // HERE BE DRAGONS Make sure you're using the correct matrix for the grayscale and/or color cameras!
       1161:   ReadKittiOdometryCalibration(dataset_root + "/" + input_config.calibration_fname,
       1162:                                left_gray_proj, right_gray_proj, left_color_proj, right_color_proj,
      >1163:                                velo_to_left_gray_cam, downscale_factor);
       1164: 
       1165:   Eigen::Vector2i frame_size = GetFrameSize(dataset_root, input_config);
#6    Source "/data/code/DynSLAM/src/DynSLAM/DynSLAMGUI.cpp", line 1061, in dynslam::ReadKittiOdometryCalibration(std::string const&, Eigen::Matrix<double, 3, 4, 0, 3, 4>&, Eigen::Matrix<double, 3, 4, 0, 3, 4>&, Eigen::Matrix<double, 3, 4, 0, 3, 4>&, Eigen::Matrix<double, 3, 4, 0, 3, 4>&, Eigen::Matrix<double, 4, 4, 0, 4, 4>&, double) [0x48ba1a]
       1058:   static const string kRightColor = "P3:";
       1059:   ifstream in(fpath);
       1060:   if (! in.is_open()) {
      >1061:     throw runtime_error(utils::Format("Could not open calibration file: [%s]", fpath.c_str()));
       1062:   }
       1063: 
       1064:   left_gray_proj = ReadProjection(kLeftGray, in, downscale_factor);
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6, at 0x7f9bb82bedf7, in __cxa_throw
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6, at 0x7f9bb82bebe0, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6, at 0x7f9bb82beb95, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6, at 0x7f9bb82c0bbc, in __gnu_cxx::__verbose_terminate_handler()
#1    Source "/build/eglibc-oGUzwX/eglibc-2.19/stdlib/abort.c", line 89, in __GI_abort [0x7f9bb7cb3027]
#0    Source "/build/eglibc-oGUzwX/eglibc-2.19/signal/../nptl/sysdeps/unix/sysv/linux/raise.c", line 56, in __GI_raise [0x7f9bb7cafc37]
Aborted (Signal sent by tkill() 25334 0)
Aborted (core dumped)
root@milton-OptiPlex-9010:/data/code/DynSLAM/build# 

```

I think that's the location inconsistency problem of calibration file, seg images, it 'll be more helpful if you could upload an example seq for testing.
Thanks!

AndreiBarsan, 2017-11-24:
Thanks for the interest in the project! 
Is that a raw KITTI sequence or a sequence from the KITTI odometry benchmark? The system cannot work with raw sequences yet because the LIDAR is in a different format (+ a couple of other small inconsistencies).

amiltonwong, 2017-11-24:
Thanks a lot, @AndreiBarsan. :)
Yes, you may just upload a small sequence and it is very helpful for testing the code.

AndreiBarsan, 2017-11-26:
I trimmed the first 100 frames from odometry sequence 06, which is one of the few I had in my laptop, removed useless stuff, archived it, and uploaded it here: http://www.cs.toronto.edu/~iab/dynslam/mini-kitti-odometry-seq-06-for-dynslam.7z 

You'll need the 7zip (un)archiver to extract it. I used it instead of zip because it provides better compression. You should be able to point `--dataset_root` to the folder where that archive is extracted and get the ball rolling.

Please let me know how it goes, and if it's good, I'll add the link to the README!

amiltonwong, 2017-11-27:
Thanks @AndreiBarsan ,

I just downloaded and extracted it.

`./DynSLAMGUI  --dataset_root=/data/DynSLAM_dataset/mini-kitti-odometry-seq-06-for-dynslam/mini-seq-06` 

Here is the [run log](https://gist.githubusercontent.com/amiltonwong/2278a4aa591a9b5601677857ebbf3fc6/raw/2849b3d217ea5e57be9e9914d9fcee230b097e66/run%2520log).

It throws 
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not open CSV file. Does the folder it should be in exist?
```

What should I do for this CSV file? I've checked that there's no any *.csv file located at "mini-seq-06/".

Thanks!

AndreiBarsan, 2017-11-27:
Ah, stupid mistake on my side. Try creating a 'csv' folder in the project root.

amiltonwong, 2017-11-28:
Thanks @AndreiBarsan ,
Now it works and the pangolin GUI appears showing the result :)

closed 2017-11-28 by amiltonwong