AndreiBarsan/DynSLAM #45: build error with gflags

tuchengyuan, 2017-12-01:
I got a fatal build error as the following:

~/DynSLAM/build$ make
Scanning dependencies of target InstRecLib
[  0%] Building CXX object src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/SegmentationDataset.cpp.o
[  1%] Building CXX object src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/SparseSFProvider.cpp.o
[  2%] Building CXX object src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/InstanceTracker.cpp.o
In file included from /home/tutu/DynSLAM/src/DynSLAM/InstRecLib/Track.h:7:0,
                 from /home/tutu/DynSLAM/src/DynSLAM/InstRecLib/InstanceTracker.h:13,
                 from /home/tutu/DynSLAM/src/DynSLAM/InstRecLib/InstanceTracker.cpp:3:
/home/tutu/DynSLAM/src/DynSLAM/InstRecLib/../InfiniTamDriver.h:11:27: fatal error: gflags/gflags.h: No such file or directory
compilation terminated.
src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/build.make:110: recipe for target 'src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/InstanceTracker.cpp.o' failed
make[2]: *** [src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/InstanceTracker.cpp.o] Error 1
CMakeFiles/Makefile2:1247: recipe for target 'src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/all' failed
make[1]: *** [src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

AndreiBarsan, 2017-12-01:
Try going to `src/gflags`and building that using CMake. It should get placed in the CMake registry and found by CMake.

tuchengyuan, 2017-12-01:
Thanks! But I encounter another build error after building the gflags

cd ~/DynSLAM/src/gflags
mkdir build && cd build
export CXXFLAGS="-fPIC" && cmake .. && make VERBOSE=1
sudo make install

cd ~/DynSLAM/build
build

[  1%] Building CXX object src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/InstanceTracker.cpp.o
[  1%] Building CXX object src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/PrecomputedSegmentationProvider.cpp.o
[  2%] Building CXX object src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/InstanceReconstructor.cpp.o
[  3%] Building CXX object src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/VisoSparseSFProvider.cpp.o
/home/tutu/DynSLAM/src/DynSLAM/InstRecLib/VisoSparseSFProvider.cpp: In function ‘Eigen::Matrix4d instreclib::VisoToEigen(const Matrix&)’:
/home/tutu/DynSLAM/src/DynSLAM/InstRecLib/VisoSparseSFProvider.cpp:9:28: error: passing ‘const Matrix’ as ‘this’ argument discards qualifiers [-fpermissive]
   return Eigen::Matrix4d((~viso_matrix).val[0]);
                            ^
In file included from /home/tutu/DynSLAM/src/DynSLAM/InstRecLib/../../libviso2/src/matcher.h:34:0,
                 from /home/tutu/DynSLAM/src/DynSLAM/InstRecLib/SparseSFProvider.h:7,
                 from /home/tutu/DynSLAM/src/DynSLAM/InstRecLib/VisoSparseSFProvider.h:7,
                 from /home/tutu/DynSLAM/src/DynSLAM/InstRecLib/VisoSparseSFProvider.cpp:3:
/home/tutu/DynSLAM/src/DynSLAM/InstRecLib/../../libviso2/src/matrix.h:106:11: note:   in call to ‘Matrix Matrix::operator~()’
   Matrix  operator~ ();                // transpose
           ^
src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/build.make:182: recipe for target 'src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/VisoSparseSFProvider.cpp.o' failed
make[2]: *** [src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/VisoSparseSFProvider.cpp.o] Error 1
CMakeFiles/Makefile2:1247: recipe for target 'src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/all' failed
make[1]: *** [src/DynSLAM/InstRecLib/CMakeFiles/InstRecLib.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

AndreiBarsan, 2017-12-03:
Hmm, try https://github.com/AndreiBarsan/DynSLAM/issues/33 as it seems like a similar problem. (I don't have access to a CUDA-enabled workstation to test these things at the moment. I'm sorry about that... :( )

tuchengyuan, 2017-12-04:
The solution on #33 works. I modify VisoSparseSFProvider.cpp as the following:
Eigen::Matrix4d VisoToEigen(const Matrix &viso_matrix) {
  //  The '~' transposes the matrix...
  Matrix a=viso_matrix;
  return Eigen::Matrix4d((~a).val[0]);
}

Then make can go on until meet the the following link error:

[ 65%] Linking CXX executable DynSLAMGUI
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DynSLAMGUI.cpp.o: In function `backward::TraceResolverLinuxImpl<backward::trace_resolver_tag::libdw>::find_fundie_by_pc(Dwarf_Die*, unsigned long, Dwarf_Die*)':
/home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp/backward.hpp:1394: undefined reference to `dwarf_child'
/home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp/backward.hpp:1409: undefined reference to `dwarf_attr'
/home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp/backward.hpp:1410: undefined reference to `dwarf_formflag'
/home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp/backward.hpp:1422: undefined reference to `dwarf_siblingof'
.....
.....
/home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp/backward.hpp:1373: undefined reference to `dwarf_formudata'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DynSLAMGUI.cpp.o: In function `void backward::details::deleter<void, Dwfl*, &dwfl_end>::operator()<Dwfl*>(Dwfl*&) const':
/home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp/backward.hpp:303: undefined reference to `dwfl_end'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DynSLAMGUI.cpp.o: In function `backward::TraceResolverLinuxImpl<backward::trace_resolver_tag::libdw>::die_has_pc(Dwarf_Die*, unsigned long)':
/home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp/backward.hpp:1371: undefined reference to `dwarf_attr'
/home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp/backward.hpp:1373: undefined reference to `dwarf_formudata'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DynSLAMGUI.cpp.o: In function `void backward::details::deleter<void, Dwfl*, &dwfl_end>::operator()<Dwfl*>(Dwfl*&) const':
/home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp/backward.hpp:303: undefined reference to `dwfl_end'
collect2: error: ld returned 1 exit status
CMakeFiles/DynSLAMGUI.dir/build.make:821: recipe for target 'DynSLAMGUI' failed
make[2]: *** [DynSLAMGUI] Error 1
CMakeFiles/Makefile2:74: recipe for target 'CMakeFiles/DynSLAMGUI.dir/all' failed
make[1]: *** [CMakeFiles/DynSLAMGUI.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

the cmake result is the following:
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- backwards-cpp-enhanced stack traces? ON
-- Found Eigen3: /home/tutu/DynSLAM/src/eigen (Required is at least version "2.91.0") 
-- Found GLUT: /usr/lib/x86_64-linux-gnu/libglut.so  
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- CUDA found: TRUE
-- Found libdw: /usr/lib/x86_64-linux-gnu/libdw.so  
-- Found libbfd: /usr/lib/x86_64-linux-gnu/libbfd.so  
-- BACKWARD_HAS_UNWIND=1
-- BACKWARD_HAS_BACKTRACE=0
-- BACKWARD_HAS_BACKTRACE_SYMBOL=0
-- BACKWARD_HAS_DW=1
-- BACKWARD_HAS_BFD=0
-- Found Backward: /home/tutu/DynSLAM/src/InfiniTAM/InfiniTAM/ORUtils/third_party/backward-cpp  
-- Looking for C++ include unistd.h
-- Looking for C++ include unistd.h - found
-- Looking for C++ include stdint.h
-- Looking for C++ include stdint.h - found
-- Looking for C++ include inttypes.h
-- Looking for C++ include inttypes.h - found
-- Looking for C++ include sys/types.h
-- Looking for C++ include sys/types.h - found
-- Looking for C++ include sys/stat.h
-- Looking for C++ include sys/stat.h - found
-- Looking for C++ include fnmatch.h
-- Looking for C++ include fnmatch.h - found
-- Looking for C++ include stddef.h
-- Looking for C++ include stddef.h - found
-- Check size of uint32_t
-- Check size of uint32_t - done
-- Looking for strtoll
-- Looking for strtoll - found
-- CUDA found: TRUE
-- OpenNI found: FALSE
-- LibRoyale found: FALSE
-- libuvc found: FALSE
-- librealsense found: FALSE
-- libpng found: TRUE
-- Compile: 52
-- compiled -> 0
-- worked
-- Using CUDA compute capability: 52
-- Using CUDA arch: 52
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tutu/DynSLAM/build

Before doing cmake, I apt-get install  libdw-dev binutils-dev in order to let cmake find libdw and libbfd.

So how to do next to fix the current link issue?
Thanks!

AndreiBarsan, 2017-12-04:
Hmm, interesting. Could you try to disable `libdw` support from CMake? Alternatively, you could try to add `#define BACKWARD_HAS_DW 0` to `DynSlamGUI.cpp`, but I'm not 100% sure it would work.

I only used `libbfd` during development and didn't see any error like that, so it may be due to some `libdw` weirdness.

tuchengyuan, 2017-12-05:
After executing sudo apt-get remove  libdw-dev, I make successfully at last. My whole script for building on Ubuntu 16.04 is:

sudo apt-get install git libxmu-dev libxi-dev freeglut3 freeglut3-dev glew-utils libglew-dev libglew-dbg
sudo apt-get install cmake doxygen libboost-python-dev libopencv-dev libboost-all-dev python-opencv
sudo apt-get install binutils-dev libpng++-dev

git clone --recursive https://github.com/AndreiBarsan/DynSLAM

cd DynSLAM/src/Pangolin/
mkdir build
cd build
cmake ..
make

cd ../../gflags
mkdir build && cd build
export CXXFLAGS="-fPIC" && cmake .. && make VERBOSE=1
sudo make install

nano ~/DynSLAM/src/DynSLAM/InstRecLib/VisoSparseSFProvider.cpp
    Eigen::Matrix4d VisoToEigen(const Matrix &viso_matrix) {
      //  The '~' transposes the matrix...
      Matrix a=viso_matrix;
      return Eigen::Matrix4d((~a).val[0]);
    }

cd ~/DynSLAM
mkdir build
cd build
cmake ..
make

closed 2017-12-05 by tuchengyuan

AndreiBarsan, 2017-12-09:
That's awesome news. 
I really appreciate you sharing the setup script!!

Have you managed to get the system to run on an example sequence, such as this short one? http://www.cs.toronto.edu/~iab/dynslam/mini-kitti-odometry-seq-06-for-dynslam.7z

tuchengyuan, 2017-12-10:
Yeah.  I run "build/DynSLAM --use_dispnet --dataset_root=path/to/extracted/archive" with mini-kitti-odometry-seq-06-for-dynslam.7z successfully. 
But is still on the way to test with KITTI Odometry dataset.