AndreiBarsan/DynSLAM #47: Error when compiling InstRecLib

wj4y78308, 2018-04-09:
Hi, I want to use your project but I face some problems now, and I really need help. :(

1. Before compiling DynSLAM project, I compile InstRecLib and get these errors:

/home/imlab-slam/DynSLAM-master/src/DynSLAM/InstRecLib/InstanceReconstructor.cpp: In member function ‘void instreclib::reconstruction::InstanceReconstructor::SaveObjectToMesh(int, const string&)’:
/home/imlab-slam/DynSLAM-master/src/DynSLAM/InstRecLib/InstanceReconstructor.cpp:750:30: error: expected type-specifier before ‘ITMMeshingEngine_CUDA’
   auto *meshing_engine = new ITMMeshingEngine_CUDA<ITMVoxel, ITMVoxelIndex> (
                              ^
/home/imlab-slam/DynSLAM-master/src/DynSLAM/InstRecLib/InstanceReconstructor.cpp:763:10: error: type ‘<type error>’ argument given to ‘delete’, expected pointer
   delete meshing_engine;
          ^
make[2]: *** [CMakeFiles/InstRecLib.dir/InstanceReconstructor.cpp.o] Error 1
make[1]: *** [CMakeFiles/InstRecLib.dir/all] Error 2
make: *** [all] Error 2

I think the problem is that in InstanceReconstructor.cpp, there is no include related to ITMMeshingEngine_CUDA, so I just include it like this: 

#include ITMMeshingEngine_CUDA.h

2. After include ITMMeshingEngine_CUDA.h file, I got another errors:
/home/imlab-slam/DynSLAM-master/src/DynSLAM/InstRecLib/../../InfiniTAM/InfiniTAM/ITMLib/Engine/DeviceSpecific/CUDA/ITMMeshingEngine_CUDA.h:48:9: error: ‘__global__’ does not name a type
  extern __global__ void findAllocatedBlocks(Vector4s *visibleBlockGlobalPos,

and I also search for the solution... and found that terms related to CUDA(e.g. __globla__ in ITMMeshingEngine_CUDA.h) are not allowed in .cpp files, which is a compiler issue.

Because I am a beginner of CUDA, I really don't know how to solve this problem.
Can anyone help me please? really thanks!

closed 2018-04-10 by wj4y78308

AndreiBarsan, 2018-04-11:
Hi wj4y78308, and thanks for checking out the project!

I believe that the reason you are seeing the second error is because you are including a CUDA-specific implementation in a cpp file which should be platform-agnostic.

The first error looks like a compiler version mismatch. What version of g++ are you using, and what operating system are you on?

wj4y78308, 2018-04-29:
Thank you for replying me!
I alreadly solved this problem :)

AndreiBarsan, 2018-04-29:
That's great to hear! Can you describe your solution briefly, in case someone else experiences a similar problem in the future? Thank you!