AndreiBarsan/DynSLAM #52: ORUtils/MemoryBlock.h(239) : cudaSafeCall() Runtime API error : 30 | unknown error.

wuxiaoqiang12, 2018-08-01:
When I run your program, I encountered a problem that like the title description. I don't know how to fix it.So I need your help. Thanks a lot.

wuxiaoqiang12, 2018-08-01:
➜  DynSLAM git:(master) ✗ build/DynSLAMGUI --dataset_root=/home/jon/dataset/Kitti/mini-seq-06
Read calibration from KITTI-style data...
Frame size: 1226
 370
Proj (left, gray): 
707.091       0 601.887       0
      0 707.091  183.11       0
      0       0       1       0
Proj (right, gray): 
 707.091        0  601.887 -379.815
       0  707.091   183.11        0
       0        0        1        0
Proj (left, color): 
   707.091          0    601.887    46.8878
         0    707.091     183.11    0.11786
         0          0          1 0.00620322
Proj (right, color): 
  707.091         0   601.887   -333.46
        0   707.091    183.11   1.93013
        0         0         1 0.0033185
Velo: 
-0.00185774   -0.999966 -0.00803998 -0.00478403
-0.00648147  0.00805186   -0.999947  -0.0733743
   0.999977 -0.00180553  -0.0064962   -0.333997
          0           0           0           1
/home/jon/Applications/DynSLAM/src/InfiniTAM/InfiniTAM/ITMLib/Engine/../Utils/../Objects/../../ORUtils/MemoryBlock.h(239) : cudaSafeCall() Runtime API error : 30 | unknown error.

AndreiBarsan, 2019-06-09:
I can't seem to reproduce this. If anyone else encounters this issue, could you please update the thread with additional info like GPU model, driver version, CUDA version, OS, etc.?

wuxiaoqiang12, 2019-06-11:
Sorry, I cannot provide more info about this issue, because now I focus on
another field(formal verification). I remember this issue occurs when you
have a low memory of GPU. It's lower than the program required.

AndreiBarsan, 2019-06-14:
@wuxiaoqiang12 Thank you for taking the time to provide the clarification! I'm sure this will be helpful to people in the future if they have similar problems.

closed 2019-06-14 by AndreiBarsan