AndreiBarsan/DynSLAM #55: can't clone due to dependencies on non-public eigen repository

catphive, 2019-03-13:
When trying to do:
git clone --recursive https://github.com/AndreiBarsan/DynSLAM

It fails to download because the eigen submodule on github is either private or no longer exists:

fatal: Authentication failed for 'https://github.com/RLovelett/eigen/'

Is there something special about this eigen repository? Can we just replace it with any eigen mirror on github?

AndreiBarsan, 2019-03-13:
No. I don't think there's anything special about that mirror. No idea why it's there..
I'll update the repo, but in the meantime could you try, e.g.: https://github.com/eigenteam/eigen-git-mirror/releases/tag/3.3.7 ?

AndreiBarsan, 2020-05-01:
Seems like this was due to someone making their repo private. I fixed the dependency to a public mirror!

closed 2020-05-01 by AndreiBarsan