AndreiBarsan/DynSLAM #57: Error when trying to preprocess KITTI sequences

kentsyx, 2019-04-14:
Hi, first thank you for this nice work!

I successfully built the system and tested it on mini-sequence-06, but when I try to preprocess other sequences, I first got this error
```
WARNING: Error loading config file: /home/mmvc/.docker/config.json: stat /home/mmvc/.docker/config.json: permission denied
docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.39/containers/create: dial unix /var/run/docker.sock: connect: permission denied.
See 'docker run --help'.
Failure: Could not build or run DispNet on sequence ID [2] under path: [data/kitti/tracking/training]
```
So naturally I put ```sudo``` in front of it but then I got this
```
Unable to find image 'dispflownet:latest' locally
docker: Error response from daemon: pull access denied for dispflownet, repository does not exist or may require 'docker login'.
See 'docker run --help'.
Failure: Could not build or run DispNet on sequence ID [2] under path: [data/kitti/tracking/training]
```
I went and checked the path variables in ```preprocess-sequence.sh``` and they seem all correct. I also did ```docker login``` but no luck as well. I'm quite new to Docker stuff can not figure out what I missed. A little help would be appreciated!

kentsyx, 2019-04-14:
I'm on ubuntu 16.04.6 LTS with cuda 8.0, cudnn 6.0, docker 18.09.5 and TITAN X GPU

closed 2019-04-14 by kentsyx

AndreiBarsan, 2019-04-17:
Oh, that's great! If you have time, could you please post here about how you solved the problem in case other people face it in the future?

kentsyx, 2019-04-24:
Hi, just saw this. It was a silly mistake I forgot to build the dispnet docker at first,after I built it everything works fine

AndreiBarsan, 2019-04-30:
Thanks for clarifying! Let me know if you have further questions ;)

0dmf0, 2020-05-03:
Unable to find image 'dispflownet:latest' locally
Can you explain clearly how to solve this problem? 
Thanks

AndreiBarsan, 2020-05-03:
Yeah, just building the appropriate image should do the trick. Please see: https://github.com/AndreiBarsan/dispnet-flownet-docker/blob/master/

and the scripts in that dir! Let me know if you have issues running this but Docker should make it overall nicer to handle than DynSLAM itself 😅

0dmf0, 2020-05-04:
dmf@dmf-GS63-7RE:/media/dmf/丁/DynSLAM$ ./scripts/preprocess-sequence.sh kitti-odometry ./data/kitti/odometry/sequences/01
Running dispnet preprocessing script!
Using GPU:       0
Running network: DispNetCorr1D-K
Working dir:     /dispflownet/dispflownet-release/models/DispNetCorr1D-K
First input:     left-dispnet-in.txt
Second input:    right-dispnet-in.txt
Output:          left-dispnet-out.txt
Using default tag: latest
nvidia-docker | 2020/05/04 20:38:26 Error: Error response from daemon: pull access denied for dispflownet, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
Failure: Could not build or run DispNet on sequence ID [] under path: [./data/kitti/odometry/sequences/01/]

When I want to process a KITTI data ,this happened. Do you know why?

0dmf0, 2020-05-04:
> Yeah, just building the appropriate image should do the trick. Please see: https://github.com/AndreiBarsan/dispnet-flownet-docker/blob/master/
> 
> and the scripts in that dir! Let me know if you have issues running this but Docker should make it overall nicer to handle than DynSLAM itself sweat_smile

The website is broken. Not that page.

AndreiBarsan, 2020-05-04:
sorry try https://github.com/AndreiBarsan/dispnet-flownet-docker/