AndreiBarsan/DynSLAM #60: Compared to the original InfiniTAM, does this implemention need parameter adjustment?

MicroYY, 2019-05-17:
I am wondering that since the original InfiniTAM works indoor and DynSLAM works outdoor, how to adjust the parameters so that InfiniTAM can work outdoor? The depth information is quite different in these two situations.

AndreiBarsan, 2019-05-21:
This depends on the sensor you are using, how your stereo rig is configured, etc.
Also, in many outdoor scenarios the motion tends to be faster than in indoor ones (e.g., in KITTI you can have very fast motion but only 10Hz inputs), so you'd need to use a different camera tracking method. 

Do you have any details about the context in which you wish to make this work? Do you want to run DynSLAM on indoor data?

MicroYY, 2019-05-22:
Thanks for your reply!
I want to use a stereo camera which outputs depth map in real time for outdoor reconstruction.
I employ [VINS-MONO](https://github.com/HKUST-Aerial-Robotics/VINS-Mono) as the visual odometry, so the camera pose is well got. 
Now the InfiniTAM can reconstruct indoor scenarios quite well, but when it comes to outdoor objects, the result is not that satisfactory. My stereo camera outputs depth up to 20m, and obviously the original InfiniTAM cannot fuse depth that far.
 I knwo there are several parameters affecting the reconstrucion quality. The code defines a class called **sceneParams** which wraps the parameters related to the scene representation.  I tried several combinations of these parameters but cannot achieve a satisfactory result. 
So how do DynSLAM adjust the parameters so that it can work outdoor? Do you have a set of parameters? If the parameters are suitable, maybe the problem comes from depth map.

AndreiBarsan, 2019-05-29:
Yes, it's likely an issue with how the depth is computed. Have you tried tweaking the calibration parameters, e.g., the stuff from the sample file here: https://github.com/AndreiBarsan/DynSLAM/blob/a7e885205de9bf8e40527c834398c63ad2f90641/itm-sample-calib-from-kitti-odometry-sequence-06.txt ?

This calibration is read here in the code: https://github.com/AndreiBarsan/DynSLAM/blob/a7e885205de9bf8e40527c834398c63ad2f90641/src/DynSLAM/DynSLAMGUI.cpp#L1162

The last line in the calibration file is very important. This thread could be helpful in explaining it: https://github.com/victorprad/InfiniTAM/issues/130

AndreiBarsan, 2019-09-17:
Closing this due to lack of activity. Please don't hesitate to reach out if you still have questions/problems!

closed 2019-09-17 by AndreiBarsan