AndreiBarsan/DynSLAM #62: WARNING: No more room in VAB!!

Hansry, 2019-09-15:
Hello, thanks for your outstanding wroks, when I run this project, I met the problem such like "no more room in VAB"、“WARNING: Tried to allocate in excess list, but failed because the main VBA is full. vbaIdx = -8592” and then the system crash immediately. I wanna know how to fix this problem and I will very appreciate if you can answer my question.

AndreiBarsan, 2019-09-17:
Hi, what scene are you trying to reconstruct? What is your GPU model and how much memory does it have? Typically this happens when the reconstruction engine runs out of memory and cannot allocate scene blocks any more

AndreiBarsan, 2020-05-01:
Closing this issue since it's a bit old, but please feel free to re-open if you still have this problem! Like I said, it's likely due to insufficient GPU memory; sorry for the ugly error message.

closed 2020-05-01 by AndreiBarsan