AndreiBarsan/DynSLAM #64: Linking Error: "Undefined Reference" (Ubuntu 18.04)

Frozenheart1998, 2020-01-08:
Hi, I built the whole project in CLion. The messages are shown below:
`====================[ Build | DynSLAMGUI | Debug ]==============================
/home/lincoln/clion-2019.3.2/bin/cmake/linux/bin/cmake --build /home/lincoln/DynSLAM/cmake-build-debug --target DynSLAMGUI -- -j 4
[  3%] Built target gflags_nothreads_static
[  6%] Built target ORUtils
[ 16%] Built target viso2
[ 18%] Built target Utils
[ 30%] Built target InstRecLib
[ 70%] Built target ITMLib
[ 81%] Built target Engine
[ 81%] Linking CXX executable DynSLAMGUI
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DynSlam.cpp.o: In function `dynslam::DynSlam::SaveDynamicObject(std::string const&, std::string const&, int) const':
/home/lincoln/DynSLAM/src/DynSLAM/DynSlam.cpp:210: undefined reference to `instreclib::reconstruction::InstanceReconstructor::SaveObjectToMesh(int, std::string const&)'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/Evaluation/SegmentedCallback.cpp.o: In function `dynslam::eval::SegmentedCallback::GetPointAssociation(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)':
/home/lincoln/DynSLAM/src/DynSLAM/Evaluation/SegmentedCallback.cpp:26: undefined reference to `instreclib::segmentation::InstanceDetection::GetClassName() const'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/Evaluation/SegmentedCallback.cpp.o: In function `dynslam::eval::SegmentedCallback::GetPointAssociation(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)':
/home/lincoln/DynSLAM/src/DynSLAM/Evaluation/../InstRecLib/InstanceReconstructor.h:45: undefined reference to `instreclib::reconstruction::InstanceReconstructor::kPossiblyDynamicClassesVoc2012'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/Evaluation/SegmentedCallback.cpp.o: In function `dynslam::eval::SegmentedCallback::GetPointAssociation(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)':
/home/lincoln/DynSLAM/src/DynSLAM/Evaluation/SegmentedCallback.cpp:28: undefined reference to `instreclib::segmentation::InstanceDetection::GetClassName() const'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/Evaluation/SegmentedCallback.cpp.o: In function `dynslam::eval::SegmentedCallback::GetPointAssociation(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)':
/home/lincoln/DynSLAM/src/DynSLAM/Evaluation/../InstRecLib/InstanceReconstructor.h:38: undefined reference to `instreclib::reconstruction::InstanceReconstructor::kClassesToReconstructVoc2012'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DynSLAMGUI.cpp.o: In function `dynslam::gui::PangolinGui::DrawInstanceLables()':
/home/lincoln/DynSLAM/src/DynSLAM/DynSLAMGUI.cpp:507: undefined reference to `instreclib::segmentation::InstanceDetection::GetClassName() const'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DynSLAMGUI.cpp.o: In function `__static_initialization_and_destruction_0(int, int) [clone .constprop.1191]':
/home/lincoln/DynSLAM/src/DynSLAM/InstRecLib/SegmentationDataset.h:24: undefined reference to `instreclib::segmentation::labels_to_id_map(std::vector<std::string, std::allocator<std::string> >)'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DynSLAMGUI.cpp.o: In function `__static_initialization_and_destruction_0(int, int) [clone .constprop.1191]':
/home/lincoln/DynSLAM/src/DynSLAM/DynSLAMGUI.cpp:29: undefined reference to `gflags::FlagRegisterer::FlagRegisterer<std::string>(char const*, char const*, char const*, std::string*, std::string*)'
/home/lincoln/DynSLAM/src/DynSLAM/DynSLAMGUI.cpp:33: undefined reference to `gflags::FlagRegisterer::FlagRegisterer<std::string>(char const*, char const*, char const*, std::string*, std::string*)'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/DynSLAMGUI.cpp.o: In function `main':
/home/lincoln/DynSLAM/src/DynSLAM/DynSLAMGUI.cpp:1299: undefined reference to `gflags::SetUsageMessage(std::string const&)'
src/DynSLAM/InstRecLib/libInstRecLib.a(InstanceReconstructor.cpp.o): In function `instreclib::reconstruction::InstanceReconstructor::ForceObjectCleanup(int)':
/home/lincoln/DynSLAM/src/DynSLAM/InstRecLib/InstanceReconstructor.cpp:725: undefined reference to `dynslam::utils::Format(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ...)'
src/DynSLAM/InstRecLib/libInstRecLib.a(InstanceReconstructor.cpp.o): In function `instreclib::reconstruction::InstanceReconstructor::SaveObjectToMesh(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/home/lincoln/DynSLAM/src/DynSLAM/InstRecLib/InstanceReconstructor.cpp:738: undefined reference to `dynslam::utils::Format(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ...)'
src/DynSLAM/InstRecLib/libInstRecLib.a(InstanceReconstructor.cpp.o): In function `instreclib::reconstruction::InstanceReconstructor::ProcessReconstructions(bool)':
/home/lincoln/DynSLAM/src/DynSLAM/InstRecLib/InstanceReconstructor.cpp:329: undefined reference to `dynslam::utils::Format(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ...)'
/home/lincoln/DynSLAM/src/DynSLAM/InstRecLib/InstanceReconstructor.cpp:329: undefined reference to `dynslam::utils::Tic(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/DynSLAMGUI.dir/src/DynSLAM/Evaluation/EvaluationCallback.cpp.o: In function `instreclib::segmentation::SegmentationDataset::SegmentationDataset(std::string, std::vector<std::string, std::allocator<std::string> >)':
/home/lincoln/DynSLAM/src/DynSLAM/Evaluation/../InstRecLib/SegmentationDataset.h:24: undefined reference to `instreclib::segmentation::labels_to_id_map(std::vector<std::string, std::allocator<std::string> >)'
src/DynSLAM/InstRecLib/libInstRecLib.a(PrecomputedSegmentationProvider.cpp.o): In function `instreclib::segmentation::ReadMask(std::istream&, int, int)':
/home/lincoln/DynSLAM/src/DynSLAM/InstRecLib/PrecomputedSegmentationProvider.cpp:58: undefined reference to `dynslam::utils::Format(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ...)'
src/DynSLAM/InstRecLib/libInstRecLib.a(PrecomputedSegmentationProvider.cpp.o): In function `instreclib::segmentation::PrecomputedSegmentationProvider::SegmentFrame(cv::Mat_<cv::Vec<unsigned char, 3> > const&)':
/home/lincoln/DynSLAM/src/DynSLAM/InstRecLib/PrecomputedSegmentationProvider.cpp:180: undefined reference to `dynslam::utils::Format(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ...)'
/home/lincoln/DynSLAM/src/DynSLAM/InstRecLib/PrecomputedSegmentationProvider.cpp:173: undefined reference to `dynslam::utils::Format(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ...)'
collect2: error: ld returned 1 exit status
CMakeFiles/DynSLAMGUI.dir/build.make:370: recipe for target 'DynSLAMGUI' failed
make[3]: *** [DynSLAMGUI] Error 1
CMakeFiles/Makefile2:127: recipe for target 'CMakeFiles/DynSLAMGUI.dir/all' failed
make[2]: *** [CMakeFiles/DynSLAMGUI.dir/all] Error 2
CMakeFiles/Makefile2:134: recipe for target 'CMakeFiles/DynSLAMGUI.dir/rule' failed
make[1]: *** [CMakeFiles/DynSLAMGUI.dir/rule] Error 2
Makefile:177: recipe for target 'DynSLAMGUI' failed
make: *** [DynSLAMGUI] Error 2`
I cannot figure out why linking error happens since I have installed all dependencies on my computer (Ubuntu 18.04). Could someone know how to fix it?

AndreiBarsan, 2020-05-01:
I upgraded some of the code to use more modern APIs. Can you try building again? What version of CMake do you have? I use `3.10.2` and I am able to build on Ubuntu 18.04.

Frozenheart1998, 2020-05-02:
Hi, my cmake version is also 3.10.2 on Ubuntu 18.04.

AndreiBarsan, 2020-05-02:
Can you try building without CLion? Like `cd ~/DynSLAM && mkdir build && cd build && cmake .. -DWHATEVER_OTHER_PATHS_YOU_NEED=bla && make`?

And if you've already done this, can you try pulling and doing a clean build? (`rm -rf build/` and repeat)

Frozenheart1998, 2020-05-02:
I used another computer which is Ubuntu 18.04 and cloned the whole project again. It worked. I think it is related to the version of other unknown APIs since I never changed the version of CMake. Much Appreciated!

AndreiBarsan, 2020-05-02:
Oh that's awesome to hear! The linker errors did look odd. Glad you fixed
it and please let me know if you have any other issues!

closed 2020-05-02 by AndreiBarsan